jocover/jetson-ffmpeg #134: Cannot found LIB_NVBUF?

Washington-DC, 2023-08-20:
Logs:
`-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIB_NVBUF
    linked by target "nvmpi" in directory /home/tyiot/jetpack-ffmepg/jetson-ffmpeg`

I am using Agx Orin, Jetpack version 5.12，When I was compiling, I encountered this issue。

Washington-DC, 2023-08-21:
In Jetpack 5.1, the interface has changed. How should I modify it

geoncoder, 2023-10-22:
Any update on this issue? Ran into the same problem.

mrvr, 2023-11-09:
encountered same issue on xaviour NX , jetpack 5.1.2-b104.

FunnyWii, 2023-11-13:
same issue

gjrtimmer, 2024-02-04:
same issue

WangFengtu1996, 2024-02-27:
jetson agx orin dk, same issue.

fachrinnk4869, 2024-04-25:
same issue until now

FunnyWii, 2024-04-25:
> same issue until now

I think you can ignore the option "nvmpi" and try "--enable shared --disable static" cmd.

I don't know the influence of missing "nvmpi", but my method can be compiled successfully .

fancong5201314, 2024-06-19:
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIB_NVBUF
    linked by target "nvmpi" in directory /home/fancong/jetson-ffmpeg

-- Configuring incomplete, errors occurred!
See also "/home/fancong/jetson-ffmpeg/build/CMakeFiles/CMakeOutput.log".
See also "/home/fancong/jetson-ffmpeg/build/CMakeFiles/CMakeError.log".
I have the same issue, but I don't know how to solve it.

lmw0320, 2024-06-20:
Hope any update...I also met such problem on jetson orin nx， but successfully compile on jetson xavier nx。。。
Strange..

fancong5201314, 2024-06-21:
I solved it in another github

FunnyWii, 2024-06-21:
能发个链接吗老哥

lmw0320, 2024-06-21:
> […](#)

https://blog.csdn.net/cnrgq2010/article/details/135130020     这个博客里面提到了，可以删掉CMakelist.txt中的lib_nvbuf相关的内容，就可以正常编译了。。。这个是因为Jetpack版本高于5.1.1后，好像是删除了libnvbuf_utils.so文件了。

fancong5201314, 2024-06-21:
https://github.com/Keylost/jetson-ffmpeg.git这个的issue里面有说明，最好是用完整的一套，如果直接屏蔽掉那句话，会有问题。
由于nvbuf_utils库存在于5.1.2之前的jetpack版本中，
但在此版本中被完全删除。生成不再需要此库如果是Jetpack 5.1.2及其之后的版本，则需要修改CMakeLists.txt，
注释掉 find_library（LIB_NVBUF nvbuf_utils PATHS /usr/lib/aarch64-linux-gnu/tegra

egbertn, 2024-07-12:
Problem still existing on Jetpack 6.0 and Jetson Nano Orin 8GB dev edition.
Build cuda_12.2.r12.2/

in folder ffmpeg-4.2.2 
./configure --enable-nvv4l2dec --enable-libv4l2 --enable-shared --extra-libs="-L/usr/lib/aarch64-linux-gnu/tegra -lnvbuf_utils" --extra-cflags="-I /usr/src/jetson_multimedia_api/include/"
make -j6

/usr/bin/ld: libavcodec/libavcodec.so: undefined reference to `NvBuffer2Raw'
/usr/bin/ld: libavcodec/libavcodec.so: undefined reference to `NvBufferTransform'
/usr/bin/ld: libavcodec/libavcodec.so: undefined reference to `NvBufferGetParams'
/usr/bin/ld: libavcodec/libavcodec.so: undefined reference to `NvBufferCreateEx'
/usr/bin/ld: libavcodec/libavcodec.so: undefined reference to `NvBufferDestroy'
collect2: error: ld returned 1 exit status
/usr/bin/ld: libavcodec/libavcodec.so: undefined reference to `NvBuffer2Raw'
/usr/bin/ld: libavcodec/libavcodec.so: undefined reference to `NvBufferTransform'
/usr/bin/ld: libavcodec/libavcodec.so: undefined reference to `NvBufferGetParams'
/usr/bin/ld: libavcodec/libavcodec.so: undefined reference to `NvBufferCreateEx'
/usr/bin/ld: libavcodec/libavcodec.so: undefined reference to `NvBufferDestroy'