jocover/jetson-ffmpeg #136: JetPack: 5.1.3 in jetson xaiver nx compile error with LIB_NVBUF not found

fancong5201314, 2024-06-19:
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIB_NVBUF
linked by target "nvmpi" in directory /home/fancong/jetson-ffmpeg

-- Configuring incomplete, errors occurred!
See also "/home/fancong/jetson-ffmpeg/build/CMakeFiles/CMakeOutput.log".
See also "/home/fancong/jetson-ffmpeg/build/CMakeFiles/CMakeError.log".

JetPack: 5.1.3 in jetson xaiver nx

I have the same issue, but I don't know how to solve it.

nxsf8023, 2024-06-26:
libnvbuf_utils.so is not used in jetpack5 
I modification the CMakeList.txt , del LIBNVBUF about libnvbuf_utils.so
And in JP5 nvidia replace libnvbuf_utils.so by libnvbufsurface.so and libnvbufsurftransform.so
Use --extra-libs="-L/usr/lib/aarch64-linux-gnu/tegra -lnvbufsurface -lnvbufsurftransform -lnvv4l2 -L/usr/local/lib" when configure FFMPEG

fancong5201314, 2024-06-26:
Yes, I solved it use other github, thanks!

closed 2024-06-26 by fancong5201314